projectmr2024/mr_automation #5: East Pond Inflow PM04 wiring

projectmr2024, 2024-10-30:
The East Pond Inflow PM04 needs wiring from the meter, through the East Pond Hut to the west pond hut.
The wire jacket needs to be labeled - suggest "East Inflow PM04" at the locations that it is visible.
Suggest at PM04, at East Pond Hut that it transits, and West Pond hut where it terminates.
Its expected to terminate finally on an ezeio device in the west pond hut
The individual wire pairs to be terminated on Pulse1 and Pulse2 - (colour suggestion)
Wire tbd (ethernet cable) 
Possible length ~700ft

neilh10, 2025-01-23:
@benmc622 checked and this is 1" conduit and can take Cat6 standard wire

projectmr2024, 2025-01-31:
Jan 30, @benmc622 completed wiring Ethernet 6pair cable between East Pond Inflow PM04 and West Pond Hut.
Needs label, Suggested label E.Pond Inflow PM04