projectmr2024/mr_automation #6: West Pond Inflow PM04 wiring

projectmr2024, 2024-10-30:
The West Pond Inflow PM04 needs wiring from the meter,  to the west pond hut.
The wire jacket needs to be labeled - suggest "West Inflow PM04" at the locations that it is visible.
Suggest at PM04,  and West Pond hut where it terminates.
Its expected to terminate finally on an ezeio device in the west pond hut
The individual wire pairs to be terminated on Pulse1 and Pulse2 - (colour suggestion)
Wire tbd (ethernet cable) 
Possible length ~? (shortish)

neilh10, 2025-01-23:
@benmc622 checked and this can take Cat6 standard wire

projectmr2024, 2025-01-31:
Conversation with @benmc622 this wiring is complete.
Needs label suggest W.Pond Inflow PM04

benmc622, 2025-02-01:
SKU Code :MC-C6U-CCA-GY 
Cat 6 "green Exterior sheeting" @neilh10

projectmr2024, 2025-07-02:
Basics pulsing tested with Ben M, and Effegro (hotspot) on May 23. Required that signaling be two wires, seperate from power return/ground

closed 2025-07-02 by projectmr2024