projectmr2024/mr_automation #8: East Pond gravity drain plumbing for flow management

projectmr2024, 2024-10-30:
at the West pond hut, the plumbing for the East pond gravity drain requires a flow meter, an electric on/off valve and a flow defining valve.

neilh10, 2025-01-23:
@benmc622  has ordered  the Seametrics 3" flow valve, and the 3" Banjo Valve and two manual butterfly valves.
As of today @benmc622 is reworking the plumbing in the west pond hut  per the installation docs.

projectmr2024, 2025-01-29:
East pond plumbing update 
[image]

neilh10, 2025-03-12:
Benson M tested flow configuration and with some site optomizations, with input from Seametrics Tech Support is working.
The work was to wire up the seametrics 3" flow monitor, and the Banjo control valve to a test box, and sequence them through testing and check operation 
For checking the seametrics 3", the bottom flush valve was opened to the 2nd notch, and seametrics reading and totalizing noted and then for a 5G bucket test on the ouput 
For the working electrical control the Banjo 24VDC  open/close is activated and the Gate Vale set to various values.
Gate valve 4th Notch ~ 100GPM

Due to the physical configuration of the Seametrics 3" and the display facing out into the hut, the flow was the opposite and indicated a "negative" flow.   For this configuration the pulsing doesn't work, however it can totalize as "reverse flow"
To be able to use the pulsing internal wires had to be reconfigured as follows
Turn power off - battery still present
remove green face plate
remove face plate with 3 screws, requires unscrew.

[image]

Middle down left - there are two gold posts with black wires and green heatshrink. They need to be switched
Mark the lower the green wire with a black pen - this is the factory configuration 

[image]

Then carefully pull up on the green wires, and switch them around. 
When pushing back on, they will snap action ON

[image]

Assemble screen, with power off, and battery symbols shows on bottom left

[image]

Switch on 24VDC power, and the power symbol changes to external power

[image]

With the flush valve on check that the test totalizer is detecting pulses.

[image]

Seametrics support in switching the internal flow direction from :
Justice Sibomana | Technical Support Supervisor, Seametrics – Water Solutions for Life  
( 253.872.0284 | * [Jsibomana@seametrics.com](mailto:Jsibomana@seametrics.com)

closed 2025-03-27 by projectmr2024