projectmr2024/mr_automation #9: West Pond Pump Protection

projectmr2024, 2024-12-04:
The West Pond pump 2" pipe takes water from the West Pond. However the pump requires priming, and shouldn't be run for too long if it doesn't manage to prime.
In the past year its been run continuously as the west pond level has dropped. 
In the future its proposed to be pulsed on a 6hour schedule. 
The series seametrics flow meter 4-20mA is scaled for the flow passing through the meter, and also if the  pipe if not full indicate an alarm  setting the current to 3.6mA

How to create a safety pump shutoff, if after the pump has switched ON, AND after say 5seconds the seametrics 4-20mA is not above 4mA, to do a safety override to the pump, and indicate to the controller the alarm condition.

projectmr2024, 2025-03-07:
Feb25 agreed with Joel that any algorithmic protection will be to check if the seametrics is full, and that assumes pump is primed. 
If seametrics detected as empty from 20mA, then do not activate pump, and send email.
The algorithm will do this on every occurrence, until local land managers changes the schedule.

If West Pond drains, and reaches the bottom, then seametrics becomes empty, then stops pump activation by same process.
Land Manager needs to stop pumping. 
The LTxxx gage absolute depth sensor has 4-20mA, but its pre-weather adjustment,  so effective accuracy is suspect,

For seametrics empty indication,  the land manager to  do a site visit to prime the pump.

neilh10, 2025-03-12:
Benson M tested flow configuration successfully. 
However the "empty pipe" 4-20mA configuration didn't work, probably because the instructions for the configuration wasn't feeding the power correctly.
The work was to wire up the seametrics 2" flow monitor, and the Pump 24VDC control valve to a test box, and sequence them through testing and check operation
For checking the seametrics 2",the 24VDC signal to the power isolation relay that controls the pump is operated. 
The Seametrics showed a reading of 50GPM

closed 2025-07-02 by projectmr2024

projectmr2024, 2025-07-30:
How to test?

projectmr2024, 2025-12-16:
Dec 4 Neil : At Wine Expo HotSpot booth, discussion about a demo setup they had with pump. Suggest can add a timed window from pump on to flow detection threshold. 
That is on pump turnon - if there is no flow after X (5?) minutes then shut off pump and create maintenance email.
Problem might be that this happens every 6hours