projectmr2024/mr_automation #10: WestPond Pump 24VDC

neilh10, 2025-01-23:
The relay for the westPond pump must be able to activated by 24VDC ON - enable working with [ezeio-mkii-hardware](https://ezesys.com/ezeio-mkii-hardware).
(Previously it was specified at 12VDC - needs upgrade relay)

neilh10, 2025-01-29:
Replace green unit relay with 
Relay base https://www.grainger.com/product/2DT93

Relay 1530VDC driver Mfr. Model RSL1GB4BD
Contact Rating (AC)6 A @ 250V
$15.06 https://www.grainger.com/product/SCHNEIDER-Slim-Interface-Relay-6-A-Current-55WZ42

(The wires in red I think are 120VAC and should be covered in some fashion)

[image]

projectmr2024, 2025-03-10:
March6 Ben update[](url)  - switch outside enclosure so enclosure doesn't have to be opened. 24VDC relay

[image]

projectmr2024, 2025-03-12:
BenM tested with Mar11 visit

closed 2025-03-12 by projectmr2024