projectmr2024/mr_automation #13: Test Box

projectmr2024, 2025-01-29:
The new  equipment in #7. #8, and also wiring extension #5 #6 need to be verified before the next layer of monitoring is added.
For #7, #8 - need to verify that the flow meters operate with the valves and no leakage.
This requires a box to power the flow meters, and terminate the pulse wires and count them, as well as activate the pump/relays from 24VDC and verify their operation.

neilh10, 2025-01-29:
Test unit for terminating wiring

[image]

benmc622, 2025-02-07:
Test equipment and 24v relay and din slot adapter are in my possession

projectmr2024, 2025-03-12:
BenM ran tests with Mar11 visit.  
One outstanding test is to verify Seametrics 2" PIPE EMPTY test. Required to prevent Pump burnout for condition of pipe not primed, either a leak in the pipe or empty pond.

projectmr2024, 2025-03-27:
All tests written.

closed 2025-03-27 by projectmr2024